SpudGunMan/meshing-around #12: Installation Errors

Hailo1999, 2024-07-19:
Hi There,

went to install on an Pi5 with Ubuntu 24.04 LTS

```
hailo@PI5-desktop:~/meshing-around$ ./install.sh 
[sudo] password for hailo: 

Meshing Around Installer
Do you want to install the bot in a virtual environment? (y/n)
**y**
Creating virtual environment...
./install.sh: line 21: python: command not found
./install.sh: line 22: venv/bin/activate: No such file or directory
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

```
Have on the pi Python 3.12.3 
after change in line 21 **python** to **python3** the process runs without an error

SpudGunMan, 2024-07-19:
thanks for heads up .. interesting you dont have a link for python to python3 on your pi

SpudGunMan, 2024-07-19:
fixed the line in the script

closed 2024-07-19 by SpudGunMan