SpudGunMan/meshing-around #20: Tide Error

Hailo1999, 2024-07-20:
Error handling when coordinates not available...

```
2024-07-20 13:22:56 Received DM: tide on Channel: 2 From: T-Deck (868)
System: location data for 860383800 is 47.890432,9.6894976
ERROR:root:Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1190, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 224, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number), channel_number, message_from_id)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 71, in auto_response
    tide = get_tide(str(location[0]),str(location[1]))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/modules/locationdata.py", line 53, in get_tide
    station_id = station_json['stationList'][0]['stationId']
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
TypeError: 'NoneType' object is not subscriptable

```

SpudGunMan, 2024-07-20:
Thanks had a check in the wrong spot, try a pull hope this is resolved!

Hailo1999, 2024-07-21:
no its not solved, the same error when try Tide.

try it with some german coordinates.

SpudGunMan, 2024-07-21:
I never sat down and checked this against a land locked area, and zoomed past it. Thanks for bringing attention to it :) (should be fixed now!!)

Hailo1999, 2024-07-21:
Running well, Thanks

closed 2024-07-21 by Hailo1999