SpudGunMan/meshing-around #21: Sun module Time display

Hailo1999, 2024-07-21:
If you write a `SUN` to the autoresponder, the output

```
SunRise: Mon 22 05:45
Set: Mon 22 09:10
```
is displayed
Is it possible to display sunrise and sunset times in 24 hours? Or at least that an AM and PM is employed.

SpudGunMan, 2024-07-21:
Yea I will get a 24:00 in soon

SpudGunMan, 2024-07-24:
should be resolved @Hailo1999

Hailo1999, 2024-07-26:
will try when u fix the Fetch error

Hailo1999, 2024-07-26:
I tryed Sun... ;)

```
2024-07-26 09:09:48 Received DM: Sun on Device:0 Channel: 0 From: 1128192892
ERROR:root:Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1190, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 238, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number, rxNode), channel_number, message_from_id, rxNode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 61, in auto_response
    bot_response = get_sun(str(location[0]),str(location[1]))
                               ~~~~~~~~^^^
TypeError: 'NoneType' object is not subscriptable
```

Hailo1999, 2024-07-26:
Tryed Sun over Serial connected Node Autoresponder:

```
2024-07-26 09:29:02 Received DM: sun on Device:1 Channel: 0 From: T-Deck (868)
2024-07-26 09:29:02 System: Error processing packet: 'latitude' Device:1
{'from': 860383800, 'to': 1128192892, 'decoded': {'portnum': 'TEXT_MESSAGE_APP', 'payload': b'sun', 'text': 'sun'}, 'id': 3682336776, 'rxTime': 1721978943, 'rxSnr': 5.75, 'hopLimit': 3, 'wantAck': True, 'rxRssi': -12, 'hopStart': 3, 'raw': from: 860383800
to: 1128192892
decoded {
  portnum: TEXT_MESSAGE_APP
  payload: "sun"
}
id: 3682336776
rx_time: 1721978943
rx_snr: 5.75
hop_limit: 3
want_ack: true
rx_rssi: -12
hop_start: 3
, 'fromId': '!33486a38', 'toId': '!433edb7c'}
END of packet 
```

SpudGunMan, 2024-07-26:
fixed this, thanks for this one as well!

Hailo1999, 2024-07-27:
over serial connected autoresponder it runs well but not on IP based autoresponder

SpudGunMan, 2024-07-27:
In going to close the sun issue as resolved. Can you open a new issue for the tcp issue? I will try and test using a heltec v3 in the meanwhile. but also check this out for likely all similar boards 

https://github.com/meshtastic/firmware/issues/3618

closed 2024-07-27 by SpudGunMan