SpudGunMan/meshing-around #22: Errors in replay messages

Hailo1999, 2024-07-26:
I started the autoresponder and then sent a “test” in the primary channel. then the fetch error came so that he couldn't read the other station. He doesn't respond to DM either. see error script:

Test@hailo-desktop:~/meshing-around$ ./launch.sh mesh

Meshtastic Autoresponder Bot CTL+C to exit

System: BBS Enabled, using bbsdb.pkl
System: Celestial Telemetry Enabled
System: Location Telemetry Enabled
System: Dad Jokes Enabled!
System: Store and Forward Enabled
System: Respond by DM only
2024-07-26 08:20:19 System: Autoresponder Started for Device1 Wolf Base
2024-07-26 08:21:11 Received DM: cmd on Device:0 Channel: 0 From: error fetching data
2024-07-26 08:21:12 System: Sending DM Device:0 CMD?: ping, motd, sitrep, sun, hfcond, solar, moon, tide, whereami, wx, wxc, wxa, bbslist, bbshelp, joke, messages To: error fetching data
2024-07-26 08:21:28 Received DM: sun on Device:0 Channel: 0 From: error fetching data
ERROR:root:Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1190, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 238, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number, rxNode), channel_number, message_from_id, rxNode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 61, in auto_response
    bot_response = get_sun(str(location[0]),str(location[1]))
                               ~~~~~~~~^^^
TypeError: 'NoneType' object is not subscriptable

2024-07-26 08:22:34 Ignoring DM: Hallo on Device:0 From: error fetching data
2024-07-26 08:22:34 System: Sending DM Device:0 MeshBot, here for you like a friend who is not. Try sending: ping @foo  or, cmd To: error fetching data
2024-07-26 08:22:51 Received DM: Sun on Device:0 Channel: 0 From: error fetching data
ERROR:root:Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1190, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 238, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number, rxNode), channel_number, message_from_id, rxNode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 61, in auto_response
    bot_response = get_sun(str(location[0]),str(location[1]))
                               ~~~~~~~~^^^
TypeError: 'NoneType' object is not subscriptable

SpudGunMan, 2024-07-26:
few things I didnt account for; hope to have trapped them.

also I see a problem with handling nodes, what is the config for your interface look like this is telling me I am not detecting your radio

```
2024-07-26 08:20:19 System: Autoresponder Started for Device1 Wolf Base
2024-07-26 08:21:11 Received DM: cmd on Device:0 Channel: 0 From: error fetching data
```
the device should not be 0 it should be 1

Hailo1999, 2024-07-26:
my device is a PicoPI with the interface settings see below

[interface]
# type = serial
# port = /dev/ttyACM0
# port = /dev/ttyUSB0
# port = COM1
type = tcp
hostname = 192.168.178.70
# mac = 00:11:22:33:44:55

# Additional interface for dual radio support
[interface2]
enabled = False
type = serial
port = /dev/ttyUSB0
#port = /dev/ttyACM1
# port = COM1
# hostname = 192.168.0.1
# mac = 00:11:22:33:44:55

SpudGunMan, 2024-07-26:
This should be fixed now thank you for finding these issues!

Hailo1999, 2024-07-26:
The fetch error is now gone, but I didn't get a response to my DM from the node connected via IP. The answer works great via serial.

SpudGunMan, 2024-07-26:
that is a meshtastic code API the bot just uses it .. I know TCP isnt awesome outside of MQTT on firmware

closed 2024-07-26 by SpudGunMan