SpudGunMan/meshing-around #23: Feature: look at world weather API

SpudGunMan, 2024-07-27:
https://openweathermap.org/current
Has a freemium style API 

https://open-meteo.com/en/docs
Free

Hailo1999, 2024-07-27:
with open-meteo 
`Only for non-commercial use and less than 10.000 daily API calls. See [Terms](https://open-meteo.com/en/terms) for more details.`
but i think there will not be so much access to this feature

SpudGunMan, 2024-08-07:
> open-meter has a interesting issue on pi it won't work without downloading a SDK package as well github.com/[open-meteo/open-meteo/issues/766](https://github.com/open-meteo/open-meteo/issues/766)

this issue appears to now be resolved! 👏

SpudGunMan, 2024-08-08:
try out the feature with a `git pull && git checkout worldwx`

closed 2024-08-08 by SpudGunMan