SpudGunMan/meshing-around #24: TEST over Channel 0

Hailo1999, 2024-07-27:
i try a TEST to channel LogFast, i geht the Error:

```
2024-07-27 17:58:06 System: Error processing packet: 'rxSnr' Device:1
{'from': 2803975977, 'to': 4294967295, 'decoded': {'portnum': 'TEXT_MESSAGE_APP', 'payload': b'Test', 'text': 'Test'}, 'id': 2376770240, 'rxTime': 1722095886, 'hopLimit': 3, 'viaMqtt': True, 'hopStart': 3, 'raw': from: 2803975977
to: 4294967295
decoded {
  portnum: TEXT_MESSAGE_APP
  payload: "Test"
}
id: 2376770240
rx_time: 1722095886
hop_limit: 3
via_mqtt: true
hop_start: 3
, 'fromId': '!a7214729', 'toId': '^all'}
END of packet 

```

SpudGunMan, 2024-07-27:
trapped this, if you think you can recreate it to test it but I wouldn't know how to test

Hailo1999, 2024-07-27:
With a heltec v3 (FW2.4.0) connected to serial, which is the autoresponder, the default channel is LogFast, and a T-Deck (FW GUITest) with which you send a message with the content TEST in LogFast is intercepted by the responder and tried I sent a DM and got this error message.

SpudGunMan, 2024-07-27:
yea your raw packet lacked the RXsnr value .. thats outside of the bot (underlying foundation of rf and firmware)

closed 2024-07-27 by SpudGunMan