SpudGunMan/meshing-around #25: fail to respond over tcp

SpudGunMan, 2024-07-27:
Report by @Hailo1999 pico won't work well since adding 2nd radio feature 

https://github.com/meshtastic/firmware/issues/3618

Also could be the onRx function consuming the interface need to test

SpudGunMan, 2024-07-27:
if the issue was related to serial congestion this will help identify

Hailo1999, 2024-07-27:
No, the PicoPI can only be reached via IP.

SpudGunMan, 2024-07-27:
you could use the serial headers to attach, I want to diagram this some day for how-to. do you have issues still with the fix in place?

SpudGunMan, 2024-07-27:
> No, the PicoPI can only be reached via IP.
its buffering serial over TCP, the TCP stuff is buggy as far as I know

Hailo1999, 2024-07-27:
I will do the test tomorrow. To do this I have to remove the PicoPi from the mast. Then I'll test whether the error is also present with the serial interface.

SpudGunMan, 2024-07-31:
got a fix in for TCP nodes

SpudGunMan, 2024-08-01:
closing for now - appears resolved

closed 2024-08-01 by SpudGunMan