SpudGunMan/meshing-around #26: Connection error if there is a power failure at the node

Hailo1999, 2024-08-02:
Hey, so I put the Raspberry Pico Pi back into operation, the test didn't go badly, your meshing around works great via TCP. only when the Pi is switched off and switched on again. I get the following error:

Firmware rpi-pico 2.4.045d7b82

```
Set: Sat 03 20:54 To: Wolf mobil GPS
ERROR:root:Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer
2024-08-02 10:06:54 System: Error communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1431, in run
    self.function(*self.args, **self.kwargs)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 737, in callback
    self._sendToRadio(p)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 797, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1431, in run
    self.function(*self.args, **self.kwargs)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 737, in callback
    self._sendToRadio(p)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 797, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1431, in run
    self.function(*self.args, **self.kwargs)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 737, in callback
    self._sendToRadio(p)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 797, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1431, in run
    self.function(*self.args, **self.kwargs)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 737, in callback
    self._sendToRadio(p)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 797, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1431, in run
    self.function(*self.args, **self.kwargs)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 737, in callback
    self._sendToRadio(p)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 797, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
^C
2024-08-02 10:29:43 System: Closing Autoresponder

2024-08-02 10:29:43 System: Error closing: [Errno 32] Broken pipe
System: Saving bbsdb.pkl

System: Saving Updated BBS Direct Messages bbsdm.pkl
2024-08-02 10:29:43 System: BBS Messages Saved
2024-08-02 10:29:43 System: Exiting
ERROR:asyncio:Task exception was never retrieved
future: <Task finished name='Task-3' coro=<watchdog() done, defined at /home/hailo/meshing-around/modules/system.py:464> exception=NameError("name 'retryCount' is not defined")>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/modules/system.py", line 472, in watchdog
    interface1.localNode.getMetadata()
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/node.py", line 614, in getMetadata
    self.iface.waitForAckNak()
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 651, in waitForAckNak
    raise MeshInterface.MeshInterfaceError("Timed out waiting for an acknowledgment")
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for an acknowledgment

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/hailo/meshing-around/modules/system.py", line 475, in watchdog
    await retry_interface(1)
  File "/home/hailo/meshing-around/modules/system.py", line 392, in retry_interface
    retryCount += 1
    ^^^^^^^^^^
NameError: name 'retryCount' is not defined

```

SpudGunMan, 2024-08-02:
fixed a lot of the issues - but the rebooting / power failure at the node I need to think about look into more. thanks as always for finding the other issues!

SpudGunMan, 2024-08-03:
fancy auto close, let me know should be better?

SpudGunMan, 2024-08-03:
this should be resolved I have tested success with a bunch of issues now.

Hailo1999, 2024-08-03:
runs without problems. I've only unplugged the Pi for a short time, but I'll test it tomorrow with a longer period of time

```
024-08-03 16:58:13 System: Sending DM Device:1 🏓PONG, SNR:7.5 RSSI:-72 and copy: Tryt To: T-Beam (868)
2024-08-03 16:59:41 System: Error communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment
2024-08-03 16:59:50 System: Retrying interface in 15 seconds
2024-08-03 17:00:07 System: Interface1 Opened!
2024-08-03 17:00:07 Ignoring DM: Und jetzt on Device:1 From: T-Beam (868)
```

closed 2024-08-03 by Hailo1999