SpudGunMan/meshing-around #31: Errror with sitrep

Hailo1999, 2024-08-07:
I tried the sitrep command, unfortunately the error occurs again. If I'm not mistaken, we've had this before.

Hardware: Pico Pi
Firmware:  rpi-pico 2.0.4.46d7b82

```
2024-08-07 20:38:40,707 |     INFO | Device:1 Channel: 2 Received DM: Sitrep From: ABCDE 🇩🇪 (868)
ERROR:root:Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1190, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.12/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 257, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number, rxNode), channel_number, message_from_id, rxNode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 63, in auto_response
    bot_response = "Last heard:\n" + str(get_node_list(1))
                                         ^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/modules/system.py", line 188, in get_node_list
    node_list1.sort(key=lambda x: x[1], reverse=True)
TypeError: '<' not supported between instances of 'NoneType' and 'int'

2024-08-07 20:40:23,255 |     INFO | Device:1 Channel: 2 Received DM: Solar From: ABCDE 🇩🇪 (868)
2024-08-07 20:40:24,261 |     INFO | Device:1 Sending: Elevated X-ray flux
A-Index:  6
K-Index:  1
Sunspots: 248
X-Ray Flux: C3.9
Solar Flux: 270
Signal Noise: S0-S1 To: ABCDE 🇩🇪 (868)
2024-08-07 20:40:46,100 |     INFO | Device:1 Channel: 2 Received DM: Sitrep From: ABCDE 🇩🇪 (868)
ERROR:root:Unexpected error in deferred execution <class 'TypeError'>

```

SpudGunMan, 2024-08-07:
what happens if you try it again? I assume the Firmware: rpi-pico 2.0.4.46d7b82 has something to do with this as I can't understand how this happened. I am checking for nodes if not respond, how there is a none here is curious.

what happens if you try it a second time? do you have any other nodes in the area or is there nothing but the bot? @Hailo1999

Hailo1999, 2024-08-07:
The 2nd attempt leads to the error in the last line. I already have a heltec that I could connect to meshing-around, but I wanted to use the Pico Pi for this. I can also flash the Pico Pi back to 2.3.15. I have a little more time to test again on Friday

tavdog, 2024-08-07:
I'm seeing this too on esp32 station-g1

tavdog, 2024-08-07:
then just started working randomly.

SpudGunMan, 2024-08-07:
> then just started working randomly.

@tavdog are you running official firmware? if so I assume your node was so new, I will try and trap this better

SpudGunMan, 2024-08-07:
> The 2nd attempt leads to the error in the last line. I already have a heltec that I could connect to meshing-around, but I wanted to use the Pico Pi for this. I can also flash the Pico Pi back to 2.3.15. I have a little more time to test again on Friday

@Hailo1999 I would be interested to know if updating firmware resolves this

SpudGunMan, 2024-08-07:
patched for better errors for this

SpudGunMan, 2024-08-07:
@Hailo1999 and @tavdog  if you can pull the latest, and if by chance you can recreate the error would like to see what its putting in the node table

tavdog, 2024-08-08:
I deleted the pkl file and reset node db. This is the error . Still seemed to work though.
```
2024-08-07 16:25:16,939 |     INFO | Device:1 Channel: 0 Received DM: Lheard From: Tavis Kulamalu
2024-08-07 16:25:16,939 |    ERROR | System: Error sorting node list: '<' not supported between instances of 'NoneType' and 'int'
Node List1: [('Meshtastic a106', 0, -11.75), ('Tavis Kulamalu', None, 5.5), ('Wailuku Repeater', 0, 0), ('Haleakala Repeater', 0, 0), ('Meshtastic cdd0', 0, 6.25)]

Node List2: []

2024-08-07 16:25:17,642 |     INFO | Device:1 Sending DM: Last heard:
Meshtastic a106 SNR:-11.75
Tavis Kulamalu SNR:5.5
Wailuku Repeater SNR:0
Ch Use: 13.43% To: Tavis Kulamalu
```

SpudGunMan, 2024-08-08:
thank you @tavdog appears it has a value of None for new nodes with no last heard value.

closed 2024-08-08 by SpudGunMan