SpudGunMan/meshing-around #44: Error Whereami when no GPS data set in Node

Hailo1999, 2024-08-18:
T-beam has no gps-Fix so i get an Error

Firmware: v2.3.15.deb7c27

Meshing-around Node: Pico Pi 

```
2024-08-18 10:01:42,502 |     INFO | Device:1 Channel: 0 Received DM: whereami From: T-Beam (868)
2024-08-18 10:01:42,502 |    DEBUG | System: Bot detected Commands:[{'cmd': 'whereami', 'index': 0}]
2024-08-18 10:01:42,503 |    DEBUG | System: location data for 123456789 is 0.0032768,0.0032768
ERROR:root:Unexpected error in deferred execution <class 'AttributeError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/util.py", line 288, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1297, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 301, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number, rxNode), channel_number, message_from_id, rxNode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 55, in auto_response
    bot_response = command_handler[cmds[0]['cmd']]()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 38, in <lambda>
    "whereami": lambda: handle_whereami(message_from_id, deviceID, channel_number),
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 179, in handle_whereami
    return where_am_i(str(location[0]), str(location[1]))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/modules/locationdata.py", line 36, in where_am_i
    address = location.raw['address']
              ^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'raw'

```

SpudGunMan, 2024-08-18:
should be solved with https://github.com/SpudGunMan/meshing-around/commit/8e34925af793534d18c92136c5a694404d5cdd98

Hailo1999, 2024-08-18:
again with t-Beam

```
024-08-18 11:36:45,112 |    DEBUG | System: location data for 123456789 is 0.0032768,0.0032768
ERROR:root:Unexpected error in deferred execution <class 'ValueError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/util.py", line 288, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1297, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 299, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number, rxNode), channel_number, message_from_id, rxNode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 55, in auto_response
    bot_response = command_handler[cmds[0]['cmd']]()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 38, in <lambda>
    "whereami": lambda: handle_whereami(message_from_id, deviceID, channel_number),
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 177, in handle_whereami
    return where_am_i(str(location[0]), str(location[1]))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/modules/locationdata.py", line 18, in where_am_i
    if int(lat) == 0 and int(lon) == 0:
       ^^^^^^^^
ValueError: invalid literal for int() with base 10: '0.0032768'
```

and with Heltec without a GPS atached

```
2024-08-18 11:39:55,634 |     INFO | Device:1 Channel: 0 Received DM: Whereami From: Test Node 🇩🇪 (868)
2024-08-18 11:39:55,634 |    DEBUG | System: Bot detected Commands:[{'cmd': 'whereami', 'index': 0}]
2024-08-18 11:39:55,634 |  WARNING | System: No location data for 1212121212 using default location
ERROR:root:Unexpected error in deferred execution <class 'ValueError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/util.py", line 288, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1297, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 299, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number, rxNode), channel_number, message_from_id, rxNode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 55, in auto_response
    bot_response = command_handler[cmds[0]['cmd']]()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 38, in <lambda>
    "whereami": lambda: handle_whereami(message_from_id, deviceID, channel_number),
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 177, in handle_whereami
    return where_am_i(str(location[0]), str(location[1]))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/modules/locationdata.py", line 18, in where_am_i
    if int(lat) == 0 and int(lon) == 0:
       ^^^^^^^^
ValueError: invalid literal for int() with base 10: '47.9'
```

SpudGunMan, 2024-08-18:
ahh shoot sorry, a new patch in(float)

Hailo1999, 2024-08-18:
anyway 

```
024-08-18 18:24:22,349 |     INFO | Device:1 Channel: 0 Received DM: whereami From: T-Beam (868)
2024-08-18 18:24:22,349 |    DEBUG | System: Bot detected Commands:[{'cmd': 'whereami', 'index': 0}]
2024-08-18 18:24:22,349 |    DEBUG | System: location data for 123456789 is 0.0032768,0.0032768
ERROR:root:Unexpected error in deferred execution <class 'AttributeError'>
Traceback (most recent call last):
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/util.py", line 288, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1297, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 301, in onReceive
    send_message(auto_response(message_string, snr, rssi, hop, message_from_id, channel_number, rxNode), channel_number, message_from_id, rxNode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 55, in auto_response
    bot_response = command_handler[cmds[0]['cmd']]()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 38, in <lambda>
    "whereami": lambda: handle_whereami(message_from_id, deviceID, channel_number),
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/mesh_bot.py", line 179, in handle_whereami
    return where_am_i(str(location[0]), str(location[1]))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hailo/meshing-around/modules/locationdata.py", line 36, in where_am_i
    address = location.raw['address']
              ^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'raw'

```

SpudGunMan, 2024-08-18:
a value of int(float(lat)) should have returned zero in this specific case (its interesting the value is returned like it is) 

are you on the main branch? I didnt pull this into heartbeat yet

Hailo1999, 2024-08-18:
```
raspberrypi:~/meshing-around $ git branch 
* heartbeat
  main
raspberrypi:~/meshing-around $
```

SpudGunMan, 2024-08-18:
I updated heartbeat to match main, run a `git pull` it should fix it

Hailo1999, 2024-08-18:
Perfekt

```
2024-08-18 19:53:55,804 |     INFO | Device:1 Channel: 0 Received DM: whereami From: T-Beam (868)
2024-08-18 19:53:55,839 |    DEBUG | System: Bot detected Commands:[{'cmd': 'whereami', 'index': 0}]
2024-08-18 19:53:55,839 |    DEBUG | System: location data for 3234011752 is 0.0032768,0.0032768
2024-08-18 19:53:55,839 |    ERROR | Location: No GPS data, try sending location
2024-08-18 19:53:56,540 |     INFO | Device:1 Sending DM: No location data: does your device have GPS? To: T-Beam (868)
```

closed 2024-08-18 by Hailo1999