SpudGunMan/meshing-around #48: Welcome Message new line  (\n)  error

Hailo1999, 2024-08-18:
The dm is not one of the cmd's only a text.  

```
2024-08-18 20:07:23,403 |  WARNING | Device:1 Ignoring DM: jhadsf From: T-Beam (868)
2024-08-18 20:07:23,403 |     INFO | Device:1 Sending DM: ~< Base >~   leider ist keiner anwesend   versuch es mit: ping @foo oder, cmd To: T-Beam (868)
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.11/logging/__init__.py", line 1113, in emit
    stream.write(msg + self.terminator)
UnicodeEncodeError: 'latin-1' codec can't encode characters in position 56-57: ordinal not in range(256)
Call stack:
  File "/usr/lib/python3.11/threading.py", line 995, in _bootstrap
    self._bootstrap_inner()
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/util.py", line 288, in _run
    o()
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1297, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/hailo/meshing-around/venv/lib/python3.11/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/hailo/meshing-around/mesh_bot.py", line 304, in onReceive
    msgLogger.info(f"Device:{rxNode} Channel:{channel_number} | {get_name_from_number(message_from_id, 'long', rxNode)} | " + message_string.replace('\n', '-nl-'))
Message: 'Device:1 Channel:0 | T-Beam (868) | jhadsf'
Arguments: ()
```
entry in config.ini

`welcome_message = ~< Base >~ \n leider ist keiner anwesend \n versuch es mit: ping @foo oder, cmd`

Hailo1999, 2024-08-18:
the error is since i've changed the Pi language to US

Hailo1999, 2024-08-19:
The error is still present even after removing the \n in message

SpudGunMan, 2024-08-20:
what was is the output of `echo $LANG` on a command line/shell

Hailo1999, 2024-08-20:
```
hailo@raspberrypi:~ $ echo $LANG
en_US

```

SpudGunMan, 2024-08-20:
I assume the issue is likely a local encoding issue, I default on UTF8. I would guess the switch to and from Latin1 something is unset in your OS now.

The encoding problem is in the module for logger specifically and its external to this project. The log file write is where it crashed on the bot

Hailo1999, 2024-08-21:
i chaned ihe char set to **UTF8** an now ist good, was **ISO**... was wondering why the emoji was not displayed

closed 2024-08-21 by Hailo1999