SpudGunMan/meshing-around #54: Issues with Weather Outside of US

AngryKiwiNZ, 2024-09-02:
Hey Team - Awesome wee BBS app, loving it!

When setting up for weather outside of the US this causes the app to fail to start.

# use Open-Meteo API for weather data not NOAA useful for non US locations
UseMeteoWxAPI = True

I'm running this within a VENV and the following is output:

root@OctoPi:~/meshing-around# ./launch.sh mesh
Traceback (most recent call last):
  File "/root/meshing-around/venv/lib/python3.9/site-packages/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/root/meshing-around/venv/lib/python3.9/site-packages/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/root/meshing-around/venv/lib/python3.9/site-packages/numpy/core/overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/meshing-around/venv/lib/python3.9/site-packages/numpy/__init__.py", line 130, in <module>
    from numpy.__config__ import show as show_config
  File "/root/meshing-around/venv/lib/python3.9/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "/root/meshing-around/venv/lib/python3.9/site-packages/numpy/core/__init__.py", line 50, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.9 from "/root/meshing-around/venv/bin/python3"
  * The NumPy version is: "1.26.4"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/meshing-around/mesh_bot.py", line 9, in <module>
    from modules.system import *
  File "/root/meshing-around/modules/system.py", line 50, in <module>
    from modules.wx_meteo import * # from the spudgunman/meshing-around repo
  File "/root/meshing-around/modules/wx_meteo.py", line 1, in <module>
    import openmeteo_requests # pip install openmeteo-requests
  File "/root/meshing-around/venv/lib/python3.9/site-packages/openmeteo_requests/__init__.py", line 3, in <module>
    from openmeteo_requests.Client import Client
  File "/root/meshing-around/venv/lib/python3.9/site-packages/openmeteo_requests/Client.py", line 8, in <module>
    from openmeteo_sdk.WeatherApiResponse import WeatherApiResponse
  File "/root/meshing-around/venv/lib/python3.9/site-packages/openmeteo_sdk/WeatherApiResponse.py", line 5, in <module>
    import flatbuffers
  File "/root/meshing-around/venv/lib/python3.9/site-packages/flatbuffers/__init__.py", line 15, in <module>
    from .builder import Builder
  File "/root/meshing-around/venv/lib/python3.9/site-packages/flatbuffers/builder.py", line 15, in <module>
    from . import number_types as N
  File "/root/meshing-around/venv/lib/python3.9/site-packages/flatbuffers/number_types.py", line 21, in <module>
    np = import_numpy()
  File "/root/meshing-around/venv/lib/python3.9/site-packages/flatbuffers/compat.py", line 66, in import_numpy
    import numpy as np
  File "/root/meshing-around/venv/lib/python3.9/site-packages/numpy/__init__.py", line 135, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

I can change the setting back to False and continue using Meshing-Around without issues but are keen to see if I can get this working thanks!

-Jay

SpudGunMan, 2024-09-03:
you have a corrupt venv or possibly bad data on the disk? these errors point to missing components

try install with the script?

AngryKiwiNZ, 2024-09-03:
I did use the script, but I do also have OctoPi on this Raspberry Pi which might be causing some issues perhaps. I did note that some versions of software being installed with the requirements file were already installed using different versions. Perhaps that's part of the issue.

I'm currently building a new Debian system on an older Intel NUC which I'll keep vanilla and try this all on again.

I noted the only issue was with the Open-Meteo API. Changing the statement back to "False" gets the BBS working again fine, but with the Weather module broken which I've left for now.

I'll report back when I have tried on a vanilla Debian box soon.

AngryKiwiNZ, 2024-09-03:
Thank you - I've done a whole new clean build on Debian and it's working just fine. I think it had issues with the versions already installed for OctoPrint. I'm sorry to have bothered you!

closed 2024-09-03 by AngryKiwiNZ