SpudGunMan/meshing-around #55: Whereami returns no data

AngryKiwiNZ, 2024-09-02:
Hey Team,

When running Whereami, I can see the data being returned on the console of app running inside VENV on my Raspberry Pi, but I don't get a message back on Meshtastic to send original node. I've tried this several times without success.

Kind regards,
- Jay

AngryKiwiNZ, 2024-09-02:
This seems to work around 50% of the time, based on further testing. Perhaps this issue can be closed?

SpudGunMan, 2024-09-03:
this sounds like it could be a RF issue?

SpudGunMan, 2024-09-03:
also try increasing your URL timeout? do you see any errors?

AngryKiwiNZ, 2024-09-03:
Thank you. I'm having issues where messages are being lost. Especially where Ollama is being used. Sometimes one message, sometimes all of the various message parts which is odd given I'm in the same room as the Raspberry-Pi with the USB connected radio!

I'm able to see the messages all printed on the console, but they don't appear to be transmitted via Lora/Meshtastic. Other users have noted the same.

On the other hand, damn you've done a beautiful job of integrating Gemma 2b into this. Wow it's awesome!

SpudGunMan, 2024-09-03:
Thankyou! It seems like you may have an issue with RF is the other nodes very close or far away? having the bot in client mute will also help things, it seems like you may just have a problem with packet delivery over air?

AngryKiwiNZ, 2024-09-03:
I've sorted out all of the issues which I've had but I'm still having issues with messages being sent by the R-Pi with it's USB connected Heltec v3. I can see then on the console, but sometimes the receiving never received it. When looking on a Meshtastic node I can't see the message in the Debug Panel either....

I have my node already set to Client Mute as it's trapped in my home office and doesn't need to participate in message routing when there is a dedicated outdoor router node.

I had the same issue when trying the BBS software from The Comms Channel, so could it be the Heltec perhaps? - I'll arrange a replacement and see if it helps.

Thanks again, this is seriously amazing, especially with the LLM on a R-Pi 4! - Total game changer. You rock.

SpudGunMan, 2024-09-03:
welcome! It could be the antenna or the power supply to it.. lots of stuff to troubleshoot for sure.. I am going to close this for now. if you get stuck reach out!

closed 2024-09-03 by SpudGunMan