SpudGunMan/meshing-around #58: Ollama AI Gemma 2B Accuracy Decreased

AngryKiwiNZ, 2024-09-05:
Hey Team,

Just an FYI that the latest Git Pull has dramatically decreased the accuracy of data coming from Gemma. Where the LLM was previously unable to provide data the Google search augmented search is pulling wrong information. An example is there are apparently 9 seasons of The Rookie a tv show which is wrong. Also when asked how many medals New Zealand win at the 2024 Paris Olympics, she said 1.

I'm unclear at this stage if the removal of words is contributing to this or if it's the Google search results. 

Great to see so much development going on with this project!

SpudGunMan, 2024-09-05:
without google results the LLM couldn't accurately say what 2024 they dont have up to date info, the LLM are not perfect at all. BUT if you liked the old version you can still go back easily try changing the line in llm.py https://github.com/SpudGunMan/meshing-around/blob/7afd6bbbe91e17f718ca51a8e3b7e1cdb7fcd19a/modules/llm.py#L14

change this to False

AngryKiwiNZ, 2024-09-06:
Thanks, I've implemented this and it seems better! - Another question if I may please. The LLM even with Google disabled doesn't seem to work without internet connectivity. Is this intentional? - I'd like to operate this in a Meshtastic Only environment where the R-Pi is offline.

SpudGunMan, 2024-09-06:
cleaned this up a lot, it was messy! Yes it should work fine offline.

fixed in https://github.com/SpudGunMan/meshing-around/commit/819c37bdec401eeb7363a8ba84da96af02c4ea04

closed 2024-09-08 by SpudGunMan