SpudGunMan/meshing-around #70: LemonStand throws ValueError: Currency formatting is not possible using the 'C' locale.

AngryKiwiNZ, 2024-09-30:
Hey Team,

When trying to run LemonStand within a Docker environment on top of Alpine Linux it and receive the following each time I try to run LemonStand.

 INFO | Device:1 Channel: 0 Received DM: lemonstand From: ZL4JAY Mobile
meshing-around-meshing-around-1  | 2024-09-30 19:54:08,322 |    DEBUG | System: Bot detected Commands:[{'cmd': 'lemonstand', 'index': 0}]
meshing-around-meshing-around-1  | 2024-09-30 19:54:08,322 |    DEBUG | System: Lemonade: New Player: 1156817043
meshing-around-meshing-around-1  | 2024-09-30 19:54:08,331 |    DEBUG | System: Lemonade: No high score table found
meshing-around-meshing-around-1  | 2024-09-30 19:54:08,332 |    DEBUG | System: Lemonade: Last Command: new
meshing-around-meshing-around-1  | ERROR:root:Unexpected error in deferred execution <class 'ValueError'>
meshing-around-meshing-around-1  | Traceback (most recent call last):
meshing-around-meshing-around-1  |   File "/usr/local/lib/python3.10/site-packages/meshtastic/util.py", line 288, in _run
meshing-around-meshing-around-1  |     o()
meshing-around-meshing-around-1  |   File "/usr/local/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1364, in <lambda>
meshing-around-meshing-around-1  |     lambda: pub.sendMessage(topic, packet=asDict, interface=self)
meshing-around-meshing-around-1  |   File "/usr/local/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
meshing-around-meshing-around-1  |     topicObj.publish(**msgData)
meshing-around-meshing-around-1  |   File "/usr/local/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 452, in publish
meshing-around-meshing-around-1  |     self.__sendMessage(msgData, topicObj, msgDataSubset)
meshing-around-meshing-around-1  |   File "/usr/local/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
meshing-around-meshing-around-1  |     listener(data, self, allData)
meshing-around-meshing-around-1  |   File "/usr/local/lib/python3.10/site-packages/pubsub/core/listener.py", line 237, in __call__
meshing-around-meshing-around-1  |     cb(**kwargs)
meshing-around-meshing-around-1  |   File "/app/mesh_bot.py", line 950, in onReceive
meshing-around-meshing-around-1  |     send_message(auto_response(message_string, snr, rssi, hop, pkiStatus, message_from_id, channel_number, rxNode, isDM), channel_number, message_from_id, rxNode)
meshing-around-meshing-around-1  |   File "/app/mesh_bot.py", line 94, in auto_response
meshing-around-meshing-around-1  |     bot_response = command_handler[cmds[0]['cmd']]()
meshing-around-meshing-around-1  |   File "/app/mesh_bot.py", line 40, in <lambda>
meshing-around-meshing-around-1  |     "lemonstand": lambda: handleLemonade(message_from_id, message),
meshing-around-meshing-around-1  |   File "/app/mesh_bot.py", line 352, in handleLemonade
meshing-around-meshing-around-1  |     msg += start_lemonade(nodeID=nodeID, message=message, celsius=False)
meshing-around-meshing-around-1  |   File "/app/modules/lemonade.py", line 294, in start_lemonade
meshing-around-meshing-around-1  |     buffer += " SupplyCost" + locale.currency(unit, grouping=True) + " a cup."
meshing-around-meshing-around-1  |   File "/usr/local/lib/python3.10/locale.py", line 273, in currency
meshing-around-meshing-around-1  |     raise ValueError("Currency formatting is not possible using "
meshing-around-meshing-around-1  | ValueError: Currency formatting is not possible using the 'C' locale.
meshing-around-meshing-around-1  |

SpudGunMan, 2024-09-30:
I have a test docker file if it's easy for you to try?

```
FROM python:3.10-slim
ENV PYTHONUNBUFFERED=1

RUN apt-get update && apt-get install -y gettext locales && rm -rf /var/lib/apt/lists/*

# Set the locale default to en_US.UTF-8
RUN sed -i -e 's/# en_US.UTF-8 UTF-8/en_US.UTF-8 UTF-8/' /etc/locale.gen && \
    dpkg-reconfigure --frontend=noninteractive locales && \
    update-locale LANG=en_US.UTF-8
ENV LANG en_US.UTF-8 

WORKDIR /app
COPY . /app
COPY requirements.txt .

RUN pip install -r requirements.txt
COPY . .

COPY config.ini /app/config.ini
COPY entrypoint.sh /app/entrypoint.sh

RUN chmod +x /app/entrypoint.sh
ENTRYPOINT ["/app/entrypoint.sh"]
```

AngryKiwiNZ, 2024-09-30:
Yeah man! - That works! Nice work :)

SpudGunMan, 2024-09-30:
closed with https://github.com/SpudGunMan/meshing-around/commit/316f1efd08edd4153f318b98de5c05248b6f634f

closed 2024-09-30 by SpudGunMan