SpudGunMan/meshing-around #71: Feedback on Open-Meteo API Weather Forecast

AngryKiwiNZ, 2024-09-30:
Hey Team,

All weather forcasts from Open-Meteo API (International, non-US) return with "Partly Cloudy". The remainder of the information is correct.

meshing-around-meshing-around-1  | 2024-09-30 20:05:10,123 |     INFO | Device:1 Sending Multi-Chunk DM: Today, Cond: Partly cloudy. High: 14C, with a low of 11C. No Precip. Wind: 14kph, gusts up to 29kph from:E.  To: ZL4JAY Mobile
meshing-around-meshing-around-1  | 2024-09-30 20:05:10,224 |     INFO | Device:1 Sending Multi-Chunk DM: Tomorrow, Cond: Partly cloudy. High: 14C, with a low of 12C. Precip: 98.0mm, in 14.0 hours. Wind: 27kph, gusts up to 51kph from:E.  To: ZL4JAY Mobile
meshing-around-meshing-around-1  | 2024-09-30 20:05:10,325 |     INFO | Device:1 Sending Multi-Chunk DM: Futurecast: Cond: Partly cloudy. High: 17C, with a low of 13C. Precip: 99.0mm, in 13.0 hours. Wind: 27kph, gusts up to 51kph from:E.  To: ZL4JAY Mobile

SpudGunMan, 2024-09-30:
I enhanced the values for clouds, which I didnt have detailed out before. added a few I had missed as well let me know if this clears up the forecast!

AngryKiwiNZ, 2024-10-01:
Weather information looks great now thank you!

meshing-around-meshing-around-1  | 2024-10-01 17:02:06,240 |     INFO | Device:1 Channel: 0 Received DM: Wx From: ZL4JAY Mobile
meshing-around-meshing-around-1  | 2024-10-01 17:02:08,143 |    DEBUG | System: Bot detected Commands:[{'cmd': 'wx', 'index': 0}]
meshing-around-meshing-around-1  | 2024-10-01 17:02:08,143 |    DEBUG | System: location data for 1156817043 is -41.3466624,173.2050944
meshing-around-meshing-around-1  | 2024-10-01 17:02:08,143 |    DEBUG | System: Bot Returning Open-Meteo API for weather metric
meshing-around-meshing-around-1  | 2024-10-01 17:02:09,129 |    DEBUG | Got wx data from Open-Meteo in b'Pacific/Auckland' b'NZDT'
meshing-around-meshing-around-1  | 2024-10-01 17:02:22,253 |    DEBUG | System: Splitting Message, Message Length: 375
meshing-around-meshing-around-1  | 2024-10-01 17:02:22,254 |     INFO | Device:1 Sending Multi-Chunk DM: Today, Cond: Overcast. High: 14C, with a low of 11C. No Precip. Wind: 15kph, gusts up to 31kph from:NE.  To: ZL4JAY Mobile
meshing-around-meshing-around-1  | 2024-10-01 17:02:22,355 |     INFO | Device:1 Sending Multi-Chunk DM: Tomorrow, Cond: Rain showers: Slight. High: 14C, with a low of 12C. Precip: 98.0mm, in 17.0 hours. Wind: 31kph, gusts up to 58kph from:NE.  To: ZL4JAY Mobile
meshing-around-meshing-around-1  | 2024-10-01 17:02:22,457 |     INFO | Device:1 Sending Multi-Chunk DM: Futurecast: Cond: Thunderstorm. High: 14C, with a low of 11C. Precip: 99.0mm, in 23.0 hours. Wind: 21kph, gusts up to 54kph from:NE.  To: ZL4JAY Mobile

SpudGunMan, 2024-10-01:
the API service tries to set the time to the GPS its something they do, do think I can override it and have a set value?

AngryKiwiNZ, 2024-10-01:
Thanks, I presume it's giving me the correct forecast when I compare where I live vs Auckland which is a long way away, so I think I'm looking at the correct local forecast :)

SpudGunMan, 2024-10-01:
When you see the, log `System: location data for `

It's asking the node for telemetry for and it got it (vs fallback when none) 

So the weather report went in for that requesting node exact location and the data including time zone is all sent back and should be matching is the expectations (and should be moderately accurate)

I do rely on the system having localization to the extent of time zone as a requirement which now corrected for docker should resolve issues (I hope)

AngryKiwiNZ, 2024-10-01:
> When you see the, log `System: location data for `
>
> It's asking the node for telemetry for and it got it (vs fallback when none)
>
> So the weather report went in for that requesting node exact [...]

Thank you again, I really appreciate it!

closed 2024-10-01 by AngryKiwiNZ