SpudGunMan/meshing-around #72: Sun Rise and Set Times

AngryKiwiNZ, 2024-09-30:
Hey Team,

Just a quick note to let you know that the sun rise and set times are reversed and have been for a long time, but worth a mention.

Sunrise is shown as PM (should be AM)
Sunset is shown as AM (should be PM)

 INFO | Device:1 Channel: 0 Received DM: Sun From: ZL4JAY Mobile
meshing-around-meshing-around-1  | 2024-09-30 20:19:43,765 |    DEBUG | System: Bot detected Commands:[{'cmd': 'sun', 'index': 0}]
meshing-around-meshing-around-1  | 2024-09-30 20:19:43,765 |    DEBUG | System: location data for 1156817043 is -41.3466624,173.2050944
meshing-around-meshing-around-1  | 2024-09-30 20:19:44,467 |     INFO | Device:1 Sending DM: SunRise: Tue 01 05:59PM Set: Wed 02 06:33AM

AngryKiwiNZ, 2024-09-30:
Thanks mate, love your work!

SpudGunMan, 2024-10-01:
is this also a docker? what is the local timezone of the docker?

I can push in a dockerfile with a configure string for local timezone if needed? here is the proposed solution so far

```
FROM python:3.10-slim
ENV PYTHONUNBUFFERED=1

RUN apt-get update && apt-get install -y gettext tzdata locales && rm -rf /var/lib/apt/lists/*

# Set the locale default to en_US.UTF-8
RUN sed -i -e 's/# en_US.UTF-8 UTF-8/en_US.UTF-8 UTF-8/' /etc/locale.gen && \
    dpkg-reconfigure --frontend=noninteractive locales && \
    update-locale LANG=en_US.UTF-8
ENV LANG en_US.UTF-8 
ENV TZ="America/Los_Angeles"

WORKDIR /app
COPY . /app
COPY requirements.txt .

RUN pip install -r requirements.txt
COPY . .

COPY config.ini /app/config.ini
COPY entrypoint.sh /app/entrypoint.sh

RUN chmod +x /app/entrypoint.sh
ENTRYPOINT ["/app/entrypoint.sh"]

```

AngryKiwiNZ, 2024-10-01:
> is this also a docker? what is the local timezone of the docker?
>
> I can push in a dockerfile with a configure string for local timezone if needed? here is the proposed solution so far
>
> ```
> FROM [...]

Hey Man - I'm based in New Zealand which is Pacific/New Zealand timezone. Beautiful part of the world ;).

AngryKiwiNZ, 2024-10-01:
Thank you! Sunrise and sunset times are perfect now!

closed 2024-10-01 by SpudGunMan