SpudGunMan/meshing-around #95: Change BBS to be hierarchical

joshbowyer, 2024-12-15:
Adding as a feature request for placekeeping. It would be really useful to have the BBS system be hierarchical/nested like other BBS systems, such as TC2. Users could create master boards, subcategories, etc. In the code, it could be handled by assigning "steps" to each descending level of the BBS hierarchy so that the system only ever needs to keep track of one lateral level per user, without having to use nests in nests.

mikecarper, 2025-02-01:
You could use a special char to make note of the hierarchical structure in this project without changing the db. Then when syncing with TC2 you can translate/encode the way it's expecting (TX); and be able to translate/decode when reading (RX). This way the menu doesn't need to change 
¬ https://www.ascii-code.com/CP1252/172 is one example that could be used when storing the hierarchical structure in this project.

SpudGunMan, 2025-02-01:
What about keeping track of where the user is in a message tree? 

My hangup is it's a complex navigation system which isn't ideal over hops, without a system for ACK which isn't eating up airtime as well it's adding a lot of weight to things which results in on air failure 

The simplicity is the differentiator

closed 2025-02-07 by SpudGunMan