SpudGunMan/meshing-around #99: Error number 11

hilaritas777, 2024-12-31:
Hi, Amazing project.  I keep getting this error:
'Initializing Interface1 [Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable'.
I'm running it on a raspberry pi.  Any help would be appreciated.

SpudGunMan, 2025-01-03:
sorry just saw this, and thanks :) is your user a member of dialup? access to tty device files?

SpudGunMan, 2025-01-03:
do you have gpsd or something using the port `lsof /dev/ttyUSB0`

mikecarper, 2025-02-01:
https://github.com/Meshtastic-Solutions/MeshCat might help with sharing the device. I'm still testing it

SpudGunMan, 2025-02-01:
A roadmap for firmware is to resolve some issues around shared API use between applications like mesh-bot and other meshtastic apps.

closed 2025-07-21 by SpudGunMan