SpudGunMan/meshing-around #109: Connection Drops When a Nearby Device is Detected

Pilotkosinus, 2025-01-07:
The connection to the device drops every time a nearby device is detected. The disconnection happens exactly 20 seconds after the log indicates that a nearby device has been found. This severely affects the stability of the communication.

`2025-01-07 19:02:42,942 |     INFO | Device:1 Sending DM: ✋ACK-ACK! SNR:7.75 RSSI:-62 To: *****Mobil

2025-01-07 19:03:22,644 |  WARNING | System: *****Mobil, 1338, ******, !433e5b5c at 49.06m is close to your location on Interface1 Accuracy is 32bits

2025-01-07 19:03:22,644 |     INFO | Device:1 Channel:9 SendingChannel: Sentry1: *****Mobil, 1338, ******, !433e5b5c at 49.06m

ERROR:root:Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer

2025-01-07 19:03:42,670 | CRITICAL | System: Lost Connection to Device {rxHost}

2025-01-07 19:04:02,667 |    ERROR | System: communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment

2025-01-07 19:04:02,667 |    ERROR | System: closing interface1: [Errno 32] Broken pipe

2025-01-07 19:04:02,668 |    DEBUG | System: Retrying interface1 in 15 seconds

2025-01-07 19:04:17,669 |    DEBUG | System: Retrying Interface1

2025-01-07 19:04:19,511 |    DEBUG | System: Interface1 Opened!
`

SpudGunMan, 2025-01-07:
Don't use channel 9 it's out of bounds it was my mistake to set as a default index 

Newest pull removed it from template

Pilotkosinus, 2025-01-07:
Thank you for the quick reply.

closed 2025-01-07 by Pilotkosinus