SpudGunMan/meshing-around #120: A few things not working e.g. alerts not working in GB , Rlist

g7kse, 2025-01-23:
Seeing as my current location is in a Red Alert at the moment I thought I'd check the alert function. So I entered...

`alert` return was a joke
`ea` no response
`ealert` no réponse

checked on UK alert website to confirm link in module and it is correct and showing alerts

So checked config.ini and set it to the following
`[location]
enabled = True
lat = 54.4879
lon = -3.5879
usemetric = True
repeaterlookup = rbook
noaaforecastduration = 4
noaaalertcount = 2
usemeteowxapi = True
riverlistdefault =
wxalertbroadcastenabled = True
wxalertbroadcastch = 2
enableextralocationwx = True
ealertbroadcastenabled = True
ealertbroadcastch = 2
ignorefematest = True
mysame = 053029,053073
enablegbalerts = True
enabledealerts = False
myregionalkeysde = 110000000000,120510000000
n2yoapikey =
satlist = 25544,7530`

anything look wonky? 

incidentally I was checking other commands and `rlist` returns `bug?Table not found`

I guess a few things are WIP but I also checked the list of commands in `mesh-bot.py` and unless I'm mistaken (fairly likely) then the don't all match up with the `cmd` menu.

Any help?

g7kse, 2025-01-23:
Yep, it gave me a dad joke. So I checked on the link in the modules and that's correct. There are currently 2 alerts live, so I assumed it would return both of them. 

Funnily enough I was going to set up the email function as well and that also returns a dad joke if I send `email`. 

I'm not really able to follow everything through but for the alerts if I go to the link in the `globalalert.py` module the URL for UK alerts is https://www.gov.uk/alerts this gives you a list of 2 current alerts (its going to be windy!) and if I go to https://www.gov.uk/alerts/current-alerts then that details out the alerts.

Have I followed that through ok?

SpudGunMan, 2025-01-24:
These Alerts are not ideal at all, I may need to pull this

SpudGunMan, 2025-01-24:
I just pushed a bug/typo fix which likely was related here

SpudGunMan, 2025-01-24:
I'm not sure this uk.gov html will be good for alerting with. 

There is no good way to handle this data and also get it crunched down to mesh in an easy manner that I see at the moment. 

I also can't geo code this data

closed 2025-02-01 by SpudGunMan