SpudGunMan/meshing-around #121: Allow for the install.sh script to have non-interactive execution using command-line arguments

mikecarper, 2025-02-01:
```shell
# Set default values
move=""
embedded=""
venv=""
rpi=""
bot=""
meshbotservice=""
emoji=""
ollama=""
gemma=""
reboot=""

# Function to prompt user only if variable is unset
prompt_if_unset() {
    local var_name=$1
    local prompt_text=$2
    local var_value=${!var_name}  # Get variable value dynamically

    if [[ -z "$var_value" ]]; then
        read -p "$prompt_text (y/n): " var_value
    fi

    # Ensure input is only "y" or "n"
    while [[ ! "$var_value" =~ ^[yYnN]$ ]]; do
        read -p "Invalid input. $prompt_text (y/n): " var_value
    done

    # Convert to lowercase to avoid case issues
    eval "$var_name=$(echo "$var_value" | tr '[:upper:]' '[:lower:]')"
}

# Parse command-line arguments using a switch-case
while [[ $# -gt 0 ]]; do
    case $1 in
        --move) move=$2 ;;
        --embedded) embedded=$2 ;;
        --venv) venv=$2 ;;
        --rpi) rpi=$2 ;;
        --bot) bot=$2 ;;
        --meshbotservice) meshbotservice=$2 ;;
        --emoji) emoji=$2 ;;
        --ollama) ollama=$2 ;;
        --gemma) gemma=$2 ;;
        --reboot) reboot=$2 ;;
        *)
            echo "Unknown option: $1"
            exit 1
            ;;
    esac
    shift 2  # Move to next argument pair
done

``` 

Example. This would be ran
`~/meshing-around/install.sh --move y`
and then this
`/opt/meshing-around/install.sh --embedded n --venv y --rpi y --bot mesh --meshbotservice y --emoji y --ollama n --gemma n --reboot n`

SpudGunMan, 2025-02-15:
I have a soft change freeze on this file for the moment till femtofox launches make sure no issues. I will try and port this in at some point in the future 🛸🛰️🔭

SpudGunMan, 2025-10-05:
this is being marked finished in lates pull.