SpudGunMan/meshing-around #123: rlist URL

g7kse, 2025-02-04:
I think I might have found out why list doesn't work outside of NA. If I use repeater book as the source then the website will return no results. Exactly the same as meshing around. But... There seem to be 2 proximity lookups

If I use [proximity 2 ](https://www.repeaterbook.com/row_repeaters/prox2_result.php?city=IO84el&lat=&long=&distance=25&Dunit=k&band%5B%5D=4&band%5B%5D=16&freq=&call=&mode%5B%5D=1&status_id=%25&order=distance_calc%2C+state_id+ASC)for a search for repeaters 25 miles from the maidenhead locator then it'll return 2 results

If I keep{grid} in the URL then I get the same "No data for your region error" message

If I substitute IO84el then I get a "No data for your region, if this is wrong report it" message but with one of the repeaters in the list.

So I guess if you're outside NA the URL is different...not handy I guess. So perhaps a bug

SpudGunMan, 2025-02-05:
Interesting will check this out

SpudGunMan, 2025-02-05:
could you try the latest pull see how it fairs

g7kse, 2025-02-05:
fixed for me @SpudGunMan thanks

closed 2025-02-05 by SpudGunMan