SpudGunMan/meshing-around #127: MeshBot (BBS) Web Dashboard - Network map is empty

mikecarper, 2025-02-05:
[image]

But the web server on the device does show the nodes

[image]

SpudGunMan, 2025-02-06:
It is only showing nodes with actual conversations with location data, because the web reporter is just scraping logs the only data in it is the log data. 

To get a location on the map, you need any location like data to populate it. you can use any location based data, or the sentinel is what it is handy for. The whole web reporter was a later bolt on and log scraper, why it's not like a lot of other dedicated map tools for meshtastic.

mikecarper, 2025-02-06:
Are you able to get the node list from the interface?

SpudGunMan, 2025-02-06:
Yes but accurately keeping a record in this current reporting engine may not be nice

It's scraping the bot logs not the meshtastic db

mikecarper, 2025-02-06:
Could you dump the node list once a hour while the bot is running? nodelist.json or something like that?

SpudGunMan, 2025-02-06:
Technically yes that could be done, what if it's limited to only the nodes which have interacted with the bot, thus allowing updated telemetry without interaction. 

What if it's limited to like 50

mikecarper, 2025-02-07:
As long as the raw nodelist.json is there and you do the filtering afterwards that'll work! Actually doing the full --info would probably be best for future use of the data in custom bots.

SpudGunMan, 2025-02-07:
I would do the filter pre delivery to the log. 

Can you expand on "would probably be best for future use of the data in custom bots."

This is a bot and I haven't chosen to map the entire mesh as there is a lot of other products dedicated to this, are you asking for a demo code how to do this in a bot your writing.

mikecarper, 2025-02-07:
It's just useful information that can't be gotten unless the bot is stopped. Not asking for a demo, just asking for the data to be collected and made available on the file system. I was playing around with socat to try and duplicate the serial connection but that ended up in failure with 2 writers to virtual tty devices.

SpudGunMan, 2025-02-15:
A note, you can run a node in MQTT and add the bot to MQTT with a Linux-native "meshtasticD" software only model. This is how a lot handle multiple softwares. Also a note meshtastic firmware has a possible solution to this issue by allowing multiple API connections in some future releases