SpudGunMan/meshing-around #128: Heltec v3

FredB1959, 2025-02-06:
Connected a Heltec V3 with Meshtastic firmware 2.5.20 via serial interface to RPi 4. Keep loosing its connection See log below:

Is there a solution for this issue?

pa8f@g8bpq:/opt/meshing-around $ sudo ./launch.sh mesh
2025-02-06 19:56:52,863 |    DEBUG | System: Initializing Interfaces
2025-02-06 19:56:54,314 |    DEBUG | System: Initalized Radio Device1 Node Number: 2086453788

Meshtastic Autoresponder Bot CTL+C to exit

2025-02-06 19:56:54,317 |     INFO | System: Autoresponder Started for Device1 BBS Boreft,bbsB. NodeID: 2086453788, !7c5cc21c
2025-02-06 19:56:54,318 |    DEBUG | System: Logging System Logs to disk
2025-02-06 19:56:54,318 |    DEBUG | System: BBS Enabled, data/bbsdb.pkl has 1 messages. Direct Mail Messages waiting: 0
2025-02-06 19:56:54,318 |    DEBUG | System: Celestial Telemetry Enabled
2025-02-06 19:56:54,318 |    DEBUG | System: Location Telemetry Enabled using Open-Meteo API
2025-02-06 19:56:54,319 |    DEBUG | System: Dad Jokes Enabled!
2025-02-06 19:56:54,319 |    DEBUG | System: Games Enabled!
2025-02-06 19:56:54,319 |    DEBUG | System: Wikipedia search Enabled
2025-02-06 19:56:54,319 |    DEBUG | System: MOTD Enabled using Thanks for using MeshBOT! Have a good day!
2025-02-06 19:56:54,319 |    DEBUG | System: Sentry Mode Enabled 100m radius reporting to channel:2
2025-02-06 19:56:54,319 |    DEBUG | System: Store and Forward Enabled using limit: 3
2025-02-06 19:56:54,319 |    DEBUG | System: Respond by DM only
2025-02-06 19:56:54,320 |    DEBUG | System: QRZ Welcome/Hello Enabled with training mode
2025-02-06 19:57:34,459 |    ERROR | System: communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment
2025-02-06 19:57:34,872 | CRITICAL | System: Lost Connection to Device /dev/ttyUSB0
2025-02-06 19:57:34,928 |    DEBUG | System: Retrying interface1 in 15 seconds
2025-02-06 19:57:49,931 |    DEBUG | System: Retrying Interface1
2025-02-06 19:57:51,388 |    DEBUG | System: Interface1 Opened!
2025-02-06 19:58:12,211 |    DEBUG | Telemetry:1 numPacketsRx:0 numPacketsRxErr:0 numPacketsTx:0 numPacketsTxErr:0 ChUtil%:3.2 AirTx%:0.7 totalNodes:20 Online:0 Uptime:21m Volt:4.2 Firmware:2.5.20.4c97351
2025-02-06 19:58:49,202 |     INFO | Device:1 Channel: 0 Received DM: Ping From: PA8F📟
2025-02-06 19:58:49,202 |    DEBUG | System: Bot detected Commands:[{'cmd': 'ping', 'index': 0}] From: PA8F📟
2025-02-06 19:58:50,403 |     INFO | Device:1 Sending DM: 🏓PONG SNR:-4.75 RSSI:-118 To: PA8F📟
2025-02-06 20:00:15,124 |    ERROR | System: communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment
2025-02-06 20:00:15,431 | CRITICAL | System: Lost Connection to Device /dev/ttyUSB0
2025-02-06 20:00:15,487 |    DEBUG | System: Retrying interface1 in 15 seconds
2025-02-06 20:00:30,491 |    DEBUG | System: Retrying Interface1
2025-02-06 20:00:31,966 |    DEBUG | System: Interface1 Opened!
2025-02-06 20:02:14,164 |    ERROR | System: communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment
2025-02-06 20:02:14,918 | CRITICAL | System: Lost Connection to Device /dev/ttyUSB0
2025-02-06 20:02:14,974 |    DEBUG | System: Retrying interface1 in 15 seconds
2025-02-06 20:02:29,977 |    DEBUG | System: Retrying Interface1
2025-02-06 20:02:31,461 |    DEBUG | System: Interface1 Opened!

FredB1959, 2025-02-06:
Downgraded Meshtastic FW to 2.5.15 and connection stays stable.

SpudGunMan, 2025-02-07:
I will check it out, I also saw .21 is out now and people commenting about heltec in alphanauts in meshtastic discord

FredB1959, 2025-02-07:
The RPi 4 has a 5 A power supply. Before I downgraded to 2.5.15 I tried a tcp connection with the heltec on his own USB power supply. Same issue of losing connection appeared. 

Also with 2.5.15 the issue still exists but with a lower frequency. Every 10 minutes in stead of two minutes.

Will try 2.5.21 later today.

pa8f@g8bpq:/opt/meshing-around $ sudo ./launch.sh mesh
2025-02-07 08:44:53,046 |    DEBUG | System: Initializing Interfaces
2025-02-07 08:44:55,217 |    DEBUG | System: Initalized Radio Device1 Node Numbe             r: 2086453788

Meshtastic Autoresponder Bot CTL+C to exit

2025-02-07 08:44:55,220 |     INFO | System: Autoresponder Started for Device1 B             BS Boreft,bbsB. NodeID: 2086453788, !7c5cc21c
2025-02-07 08:44:55,220 |    DEBUG | System: Logging System Logs to disk
2025-02-07 08:44:55,221 |    DEBUG | System: BBS Enabled, data/bbsdb.pkl has 1 m             essages. Direct Mail Messages waiting: 0
2025-02-07 08:44:55,221 |    DEBUG | System: Celestial Telemetry Enabled
2025-02-07 08:44:55,221 |    DEBUG | System: Location Telemetry Enabled using Op             en-Meteo API
2025-02-07 08:44:55,221 |    DEBUG | System: Dad Jokes Enabled!
2025-02-07 08:44:55,221 |    DEBUG | System: Games Enabled!
2025-02-07 08:44:55,222 |    DEBUG | System: Wikipedia search Enabled
2025-02-07 08:44:55,222 |    DEBUG | System: MOTD Enabled using Thanks for using              MeshBOT! Have a good day!
2025-02-07 08:44:55,222 |    DEBUG | System: Sentry Mode Enabled 100m radius rep             orting to channel:2
2025-02-07 08:44:55,222 |    DEBUG | System: Store and Forward Enabled using lim             it: 3
2025-02-07 08:44:55,222 |    DEBUG | System: Respond by DM only
2025-02-07 08:44:55,222 |    DEBUG | System: QRZ Welcome/Hello Enabled with trai             ning mode
2025-02-07 08:45:16,048 |    DEBUG | Telemetry:1 numPacketsRx:0 numPacketsRxErr:             0 numPacketsTx:0 numPacketsTxErr:0 ChUtil%:2.1 AirTx%:2.2 totalNodes:44 Online:0              Uptime:10h Volt:4.2 Firmware:2.5.18.89ebafc
2025-02-07 08:46:17,015 |    ERROR | System: communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment
2025-02-07 08:46:17,326 | CRITICAL | System: Lost Connection to Device /dev/ttyUSB0
2025-02-07 08:46:17,381 |    DEBUG | System: Retrying interface1 in 15 seconds
2025-02-07 08:46:32,384 |    DEBUG | System: Retrying Interface1
2025-02-07 08:46:34,561 |    DEBUG | System: Interface1 Opened!
2025-02-07 08:49:20,692 |  WARNING | System: PA8F📟, PA8F, 4201009156, !fa665004 at 88.67m is close to your location on Interface1 Accuracy is unknownbits
2025-02-07 08:49:20,692 |     INFO | Device:1 Channel:2 SendingChannel: Sentry1: PA8F📟, PA8F, 4201009156, !fa665004 at 88.67m
2025-02-07 08:55:36,199 |    DEBUG | Telemetry:1 numPacketsRx:1527 numPacketsRxErr:0 numPacketsTx:1033 numPacketsTxErr:0 ChUtil%:8.8 AirTx%:2.2 totalNodes:44 Online:35 Uptime:10h Volt:4.2 Firmware:2.5.18.89ebafc
2025-02-07 08:56:16,333 |    ERROR | System: communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment
2025-02-07 08:56:16,701 | CRITICAL | System: Lost Connection to Device /dev/ttyUSB0
2025-02-07 08:56:16,758 |    DEBUG | System: Retrying interface1 in 15 seconds

SpudGunMan, 2025-02-07:
I can not replicate this, try a flash erase and newer firmware. 

Do you have something else polling for the serial port or node?

FredB1959, 2025-02-07:
Okay problem solved. Installed meshing-around on another RPi. Changed config.ini. Connected te serial cable and no loss of connection. Just have to find out what the issue is on my RPi config.

closed 2025-02-07 by FredB1959