SpudGunMan/meshing-around #131: Make all bot commands configurable (ability to enable/disable specific commands)

jkeychan, 2025-02-13:
It would be ideal if we can configure via config.ini the commands the bot responds to. 

Specifically, in normal primary channel communications certain keywords trigger lots of bot traffic. For example, if someone uses the word "`messages`" in _any message_, the bot responds with the most recent history of messages seen on the channel. This can be disruptive and unexpected, especially if the bot is configured to respond to the channel and not via DM.

Suggestion would be to make each command configurable via commenting out lines or giving a 'true' or 'false' option for the commands. [Data Commands](https://github.com/SpudGunMan/meshing-around?tab=readme-ov-file#data-lookup)

SpudGunMan, 2025-02-14:
I was considering changing that command to history not messages but didn't want to make a hard change for everyone 

you can disable store_forward_enabled and it removes the command?

jkeychan, 2025-02-14:
Yeah that's a good workaround.