SpudGunMan/meshing-around #132: Try sending a DM again if it wasn't received

mikecarper, 2025-02-19:
On iOS I get an acknowledgment letting me know that the node actually got the message I sent. If that doesn't happen then could the bot try again 

Here's an example of a successful send to read message #3 but I had to do it twice because it got lost in the network 
[image]

SpudGunMan, 2025-02-19:
Yes this is a problem with meshtastic network. There isn't a good way to deliver with confirmations. 

Ack is only as good as the free airtime. Which isn't ideal at all. 

A feature like this can easily further degrade the network, it's a  good idea but I fear it would be a serious problem overall due to network architecture outside the scope of the bot. 

Additionally just because you asked don't mean it was actually revived, further complicating this is if there are hops it would further degrade the network for all.

mikecarper, 2025-02-19:
Would there be a way to log the ack back? This case was a zero hop; bot and node are in the same house. Maybe try again if hops are zero?

SpudGunMan, 2025-02-20:
you can turn on ACK in settings.ini it's not by default. but I am not processing any extra wait or hold today for a demanded response. could be interesting to record that value and consider it unsent for mail.. I need to look to see if there is good ways to handle it in the API.

SpudGunMan, 2025-02-26:
Checking to see, did enable

 # Request Acknowledgement of message OTA
wantAck = False

Solve the issue for you?

mikecarper, 2025-02-26:
[image]

Request for ACK but I don't see it in the logs

SpudGunMan, 2025-02-26:
Ahh sorry, I added confusion here yea I get ya, the wait time for the ack can be costly hangs the app. But for mail might be worth it

SpudGunMan, 2025-04-24:
currently there is not an ideal way to manage this with API/firmware that I know of