SpudGunMan/meshing-around #139: Quick Setup Install Folder Permissions

PeterTWP, 2025-03-04:
I have installed a couple times (works great BTW) using the ./install script.  

The first run after install always start with file permission errors (particularly in the logs) directory. I overcome this by chmod'ing them but I'm wondering if the script is missing a permissions step?

Just tyring to be helpful as I've noticed it on a couple fresh installs now.

SpudGunMan, 2025-03-04:
Yea what's the error or change needed? I have been trying to track this and haven't found it yet

dankarlin, 2025-04-10:
That user ran the script was the default user. Interesting I also had to change after= in the .service to After=multi-user.target in order to not get an error on reboot

SpudGunMan, 2025-04-10:
pushed a change which should fix the problem I think

dankarlin, 2025-04-11:
raspbian bookworm

SpudGunMan, 2025-04-24:
this should be resolved now

closed 2025-04-24 by SpudGunMan