SpudGunMan/meshing-around #147: test bug: always gets MQTT SNR:0 RSSI:0

mikecarper, 2025-05-16:
I changed the test command to always output this
```
    if hop == "Direct":
        msg = msg + hop + f" SNR:{snr} RSSI:{rssi}"
    else:
        msg = msg + hop + f" SNR:{snr} RSSI:{rssi}"
```

Updated the env meshtastic to 2.6.3
```
2025-05-16 16:28:50,812 |    DEBUG | Packet Received
 {'from': 1146655434, 'to': 2733207348, 'decoded': {'portnum': 'TEXT_MESSAGE_APP', 'payload': b'Test', 'text': 'Test'}, 'id': 2273165665, 'rxTime': 1747438130, 'hopLimit': 7, 'raw': from: 1146655434
to: 2733207348
decoded {
  portnum: TEXT_MESSAGE_APP
  payload: "Test"
}
id: 2273165665
rx_time: 1747438130
hop_limit: 7
, 'fromId': '!445892ca', 'toId': '!a2e96f34'} 
 END of packet 

2025-05-16 16:28:50,813 |    DEBUG | System: Packet HopDebugger: hop_away:0 hop_limit:7 hop_start:0
2025-05-16 16:28:50,814 |     INFO | Device:1 Channel: 0 Received DM: Test From: mcarper 92ca
2025-05-16 16:28:50,816 |    DEBUG | System: Bot detected Commands:[{'cmd': 'test', 'index': 0}] From: mcarper 92ca
2025-05-16 16:28:52,417 |     INFO | Device:1 req.ACK Sending DM: 🎙Testing MQTT SNR:0 RSSI:0 To: mcarper 92ca
2025-05-16 16:28:53,305 |    DEBUG | System: Packet Received on SerialInterface Interface
```

The Routing app is getting the SNR and RSSI though
```
2025-05-16 16:28:53,306 |    DEBUG | Packet Received
 {'from': 1146655434, 'to': 2733207348, 'decoded': {'portnum': 'ROUTING_APP', 'payload': b'\x18\x00', 'requestId': 1831528605, 'bitfield': 1, 'routing': {'errorReason': 'NONE', 'raw': error_reason: NONE
}}, 'id': 2571598942, 'rxTime': 1747438132, 'rxSnr': 13.25, 'hopLimit': 2, 'rxRssi': -13, 'hopStart': 2, 'nextHop': 52, 'relayNode': 202, 'raw': from: 1146655434
to: 2733207348
decoded {
  portnum: ROUTING_APP
  payload: "\030\000"
  request_id: 1831528605
  bitfield: 1
}
id: 2571598942
rx_time: 1747438132
rx_snr: 13.25
hop_limit: 2
rx_rssi: -13
hop_start: 2
next_hop: 52
relay_node: 202
, 'fromId': '!445892ca', 'toId': '!a2e96f34'} 
 END of packet 
```

Going to nuke the folder and try a fresh install

mikecarper, 2025-05-17:
Fresh install still has the issue. Open to debug ideas

SpudGunMan, 2025-05-17:
How are you sending packets? One to me looks like MQTT or injected like UDP. The fields are not mandatory and can be easily fudged as well.

mikecarper, 2025-05-17:
I'm sending them over RF short fast. I can turn off UDP and MQTT and try again; but LF is the only other device on the LAN so it's just 2 radios on SF

SpudGunMan, 2025-05-17:
So to answer you won't have the RSSI and SNR values in those transmissions and they are null and need to be account for 

```py
snr = -1
rssi = -1
if hop == "Direct":
        msg = msg + hop + f" SNR:{snr} RSSI:{rssi}"
    else:
        msg = msg + hop + f" SNR:{snr} RSSI:{rssi}"
```

mikecarper, 2025-05-17:
I'll turn off UDP and mqtt and see if RF picks it up.  It's been RF only tx from a t1000e as the unit sending out the test message

mikecarper, 2025-05-17:
From the log:
```
2025-05-16 23:16:35,753 |    DEBUG | System: Packet Received on SerialInterface Interface
 ('_localChannels', [settings {
  psk: "\001"
  uplink_enabled: true
  module_settings {
    position_precision: 17
  }
}
role: PRIMARY
, index: 1
settings {
  psk: 
  name: "PS-Mesh!"
  uplink_enabled: true
  downlink_enabled: true
  module_settings {
    position_precision: 17
  }
}
role: SECONDARY
, index: 2
settings {
  psk: 
  name: "PS-MQTT!"
  uplink_enabled: true
  downlink_enabled: true
  module_settings {
    position_precision: 17
  }
}
role: SECONDARY
, index: 3
settings {
  psk: 
  name: "gpio"
}
role: SECONDARY
, index: 4
settings {
  psk: "\001"
  name: "LongFast"
  uplink_enabled: true
  module_settings {
    position_precision: 13
  }
}
role: SECONDARY
, index: 5
settings {
}
, index: 6
settings {
}
, index: 7
settings {
}
])
 
 END of interface 

2025-05-16 23:16:35,755 |    DEBUG | Packet Received
 {'from': 1146655434, 'to': 2733207348, 'decoded': {'portnum': 'TEXT_MESSAGE_APP', 'payload': b'Test', 'text': 'Test'}, 'id': 485262905, 'rxTime': 1747462593, 'hopLimit': 7, 'raw': from: 1146655434
to: 2733207348
decoded {
  portnum: TEXT_MESSAGE_APP
  payload: "Test"
}
id: 485262905
[... 297 lines cut ...]
    "current": 10,
    "red": 233,
    "green": 111,
    "blue": 52,
    "ledState": false
  },
  "detectionSensor": {
    "minimumBroadcastSecs": 45,
    "detectionTriggerType": "LOGIC_HIGH",
    "enabled": false,
    "stateBroadcastSecs": 0,
    "sendBell": false,
    "name": "",
    "monitorPin": 0,
    "usePullup": false
  },
  "paxcounter": {
    "enabled": false,
    "paxcounterUpdateInterval": 0,
    "wifiThreshold": 0,
    "bleThreshold": 0
  },
  "version": 0
}
```

mikecarper, 2025-05-17:
Full erase and flash of the firmware and I'm back with SNR and RSSI getting reported

mikecarper, 2025-05-17:
Set some configuration options and the bug is back... so once I nail this down it'll probably be a firmware bug

mikecarper, 2025-05-20:
meshtastic --set store_forward.enabled true
meshtastic --set store_forward.is_server true

That'll trigger the bug

SpudGunMan, 2025-07-25:
Is the setup for this issue 

A node running S&F and the bot?

mikecarper, 2025-07-25:
Yes that's the setup for this bug. Infrastructure node was the idea. I'll do a pull request

SpudGunMan, 2025-10-05:
im gonna link and close this one, related in lots of ways. not sure how I will handle it yet. also see #security in mesh discord for ongoing issues.

SpudGunMan, 2025-10-05:
https://github.com/SpudGunMan/meshing-around/issues/196