SpudGunMan/meshing-around #151: JS8Call Support

tekstrand, 2025-05-30:
I think it would be really neat to support JS8Call in a similar way to the offline EAS integration - allowing specific callsigns or preables to trigger forwarding of messages from JS8Call out over the mesh.

SpudGunMan, 2025-10-28:
Example here for wsjt

https://github.com/ckuhtz/ham/tree/main/wsjtx

https://github.com/ckuhtz/ham/blob/main/mcast/recv_decode.py

SpudGunMan, 2025-10-28:
> I think it would be really neat to support JS8Call in a similar way to the offline EAS integration - allowing specific callsigns or preables to trigger forwarding of messages from JS8Call out over the mesh.

Give this a test? It's still fresh might need a refactor but I'm wondering if it detects 

https://github.com/SpudGunMan/meshing-around/pull/246