SpudGunMan/meshing-around #156: Connection lost and will not resume

zomgugoff, 2025-06-27:
Running 2.7.0 tech preview (although, has been a problem on 2.6+) on a T-Beam Supreme, connected over wifi to an AP about 20 feet away, onubstructed. The script will lose connection to the T-beam and not be able to reconnect. This is reported in the journal:

```Jun 25 23:24:19 gserver python3[43118]: 2025-06-25 23:24:19,194 |    DEBUG | Telemetry:1 numPacketsRx:48086 numPacketsRxErr:0 numPacketsTx:2986 numPacketsTxErr:0 ChUtil%:20.3 AirTx%:1.3 totalNodes:230 On>
Jun 25 23:29:38 gserver python3[43118]: ERROR:root:Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer
Jun 25 23:29:38 gserver python3[43118]: 2025-06-25 23:29:38,010 | CRITICAL | System: Lost Connection to Device 192.168.2.241
Jun 25 23:29:40 gserver python3[43118]: 2025-06-25 23:29:40,856 |    ERROR | System: communicating with interface1, trying to reconnect: Timed out waiting for an acknowledgment
Jun 25 23:29:40 gserver python3[43118]: 2025-06-25 23:29:40,856 |    ERROR | System: closing interface1: [Errno 32] Broken pipe```

After this happens, the script goes into a retry loop and never reconnects to the node:

```un 25 23:29:40 gserver python3[43118]: 2025-06-25 23:29:40,856 |    DEBUG | System: Retrying interface1 in 15 seconds
Jun 25 23:29:55 gserver python3[43118]: 2025-06-25 23:29:55,858 |    DEBUG | System: Retrying Interface1
Jun 25 23:29:55 gserver python3[43118]: 2025-06-25 23:29:55,859 |    ERROR | System: Error Opening interface1 on: [Errno 111] Connection refused
Jun 25 23:30:15 gserver python3[43118]: 2025-06-25 23:30:15,860 |    DEBUG | System: Retrying interface1 in 15 seconds
Jun 25 23:30:30 gserver python3[43118]: 2025-06-25 23:30:30,861 |    DEBUG | System: Retrying Interface1
Jun 25 23:30:30 gserver python3[43118]: 2025-06-25 23:30:30,862 |    ERROR | System: Error Opening interface1 on: [Errno 111] Connection refused
Jun 25 23:30:50 gserver python3[43118]: 2025-06-25 23:30:50,864 |    DEBUG | System: Retrying interface1 in 15 seconds
Jun 25 23:31:05 gserver python3[43118]: 2025-06-25 23:31:05,865 |    DEBUG | System: Retrying Interface1
Jun 25 23:31:05 gserver python3[43118]: 2025-06-25 23:31:05,866 |    ERROR | System: Error Opening interface1 on: [Errno 111] Connection refused
Jun 25 23:31:25 gserver python3[43118]: 2025-06-25 23:31:25,867 |    DEBUG | System: Retrying interface1 in 15 seconds```

Simply restarting the script gets it going again. Restarting the node while in this loop does not. It just keeps retrying.

SpudGunMan, 2025-06-27:
The API had a loss of pipe by client termination. I can't recover from this well (the API can't handle it I don't think) I know it's stable as my lab is WiFi but I have been tracking a issue with the  Linux-native connection I might be able to pass data into meshtastic for. 

To help situation try getting a better WiFi signal for the node they have crappy antennas normally.

zomgugoff, 2025-06-27:
The signal is strong from the onboard antenna. I will try an external antenna, though. Assuming that doesn't solve the problem, would it be possible to have the daemon reload on connection failure? I'd rather have a user session drop over the whole system just going offline without notice or intervention.

zomgugoff, 2025-06-29:
Just to update: I have installed an external antenna. The problem is persisting. I have also replaced the whole unit as I was having other issues with it. The problem persists.

The best I could to do to mitigate this was to edit system.py to exit instead of calling the retry function on connection error. I stuck mesh_bot.py in an infinite loop bash script so it will reload on exit.

SpudGunMan, 2025-07-01:
Yes dumping the API would do it, I might add a config.ini for this procedure to make it optional.

elixx, 2025-07-10:
I just started seeing this behaviour, too. I thought it was related to updated code I had pulled in, but don't see anything related. I also updated the node to the latest alpha when trying to resolve the issue, to no avail. So I'm not sure why this started happening for me, but I'm seeing the exact same behaviour. I found it strange that on both Windows and Linux the error string passed through are both os-specific variants of the "Connection refused" message, although it persists when the node has recovered and is definitely listening on :80. I was suspecting it might have been trying to reconnect to the wrong host.

SpudGunMan, 2025-07-10:
The API was never amazing for reconnecting

elixx, 2025-07-10:
By API do you mean `meshtastic.tcp_interface.TCPInterface` ?
I'm really curious to dig into the "Connection refused" message, as it's usually pretty telling that it's connecting to something with a closed port. But I have separate TCP monitoring on the node that shows that the node is listening on the port and shouldn't be sending any RST packets to new connections. Makes me wonder where the socket is reconnecting to.

SpudGunMan, 2025-07-22:
should be resolved

closed 2025-07-22 by SpudGunMan

zomgugoff, 2025-07-24:
I'm afraid it has not been resolved:

```
Jul 24 03:03:32 gserver python3[147338]: 2025-07-24 03:03:32,480 |    DEBUG | System: Retrying interface1 in 15 seconds
Jul 24 03:03:47 gserver python3[147338]: 2025-07-24 03:03:47,482 |    DEBUG | System: Retrying Interface1
Jul 24 03:03:47 gserver python3[147338]: 2025-07-24 03:03:47,482 |    ERROR | System: Error Opening interface1 on: [Errno 111] Connection refused
Jul 24 03:04:07 gserver python3[147338]: 2025-07-24 03:04:07,484 |    DEBUG | System: Retrying interface1 in 15 seconds
Jul 24 03:04:22 gserver python3[147338]: 2025-07-24 03:04:22,485 |    DEBUG | System: Retrying Interface1
Jul 24 03:04:22 gserver python3[147338]: 2025-07-24 03:04:22,485 |    ERROR | System: Error Opening interface1 on: [Errno 111] Connection refused
Jul 24 03:04:42 gserver python3[147338]: 2025-07-24 03:04:42,486 |    DEBUG | System: Retrying interface1 in 15 seconds
Jul 24 03:04:57 gserver python3[147338]: 2025-07-24 03:04:57,487 |    DEBUG | System: Retrying Interface1
Jul 24 03:04:57 gserver python3[147338]: 2025-07-24 03:04:57,488 |    ERROR | System: Error Opening interface1 on: [Errno 111] Connection refused
Jul 24 03:05:17 gserver python3[147338]: 2025-07-24 03:05:17,489 |    DEBUG | System: Retrying interface1 in 15 seconds
```

SpudGunMan, 2025-07-24:
Connection refused I can't handle. Neither can the API 

Testing is done by rebooting a node and reconnection. 

What did you do to get a connection refused.

SpudGunMan, 2025-07-24:
There is a new setting you might like 

```ini

#Do not retry enabling interface if it fails, just exit to let OS restart the bot
dont_retry_disconnect = False
`

zomgugoff, 2025-07-25:
> Connection refused I can't handle. Neither can the API
> 
> Testing is done by rebooting a node and reconnection.
> 
> What did you do to get a connection refused.

This was the same behavior as before where it happened randomly when just sitting idle.

> There is a new setting you might like
> 
> #Do not retry enabling interface if it fails, just exit to let OS restart the bot
> dont_retry_disconnect = False
> `

I didn't realize this was added. I was looking at the code for v1.7.9 and v.1.7.8 and saw the additions for v.1.7.8 were not present so I figured the fixed reboot code was just missing. I have enabled the option and I'll report how it goes.

SpudGunMan, 2025-07-26:
Welcome glad it's working!