SpudGunMan/meshing-around #168: How to properly use? Errors while running.

ThirstyThursten, 2025-07-28:
Helllo,
I'm not sure if I'm just stupid or if I'm missing some information..
So I did the quick install with the Install.sh, on my RaspberryPi 3B with Raspbian Lite.

I want to connect through BLE to a Heltec V3 and have this bot be sort of an Auto responder or just a fun thing to test distances etc.

I changed the config.ini 
I tried the `./launch.sh html5` this generated a page.
But gave some warnings?

```
./launch.sh html5
Checking log file: /opt/meshing-around/logs/meshbot.log
Consumed 54577 lines from /opt/meshing-around/logs/meshbot.log
Node1: N/A N/A N/A
Node2: N/A N/A N/A
Warning issue reading database file: [Errno 2] No such file or directory: '/opt/meshing-around/data/lemonstand_hs.pkl'
Warning issue reading database file: [Errno 2] No such file or directory: '/opt/meshing-around/data/dopewar_hs.pkl'
Warning issue reading database file: [Errno 2] No such file or directory: '/opt/meshing-around/data/blackjack_hs.pkl'
Warning issue reading database file: [Errno 2] No such file or directory: '/opt/meshing-around/data/videopoker_hs.pkl'
Warning issue reading database file: [Errno 2] No such file or directory: '/opt/meshing-around/data/mmind_hs.pkl'
Warning issue reading database file: [Errno 2] No such file or directory: '/opt/meshing-around/data/golfsim_hs.pkl'
Warning issue reading database file: [Errno 2] No such file or directory: '/opt/meshing-around/data/qrz.db'

MeshBot (BBS) Web Dashboard Report Generator

Main dashboard: file:///opt/meshing-around/etc/www/index.html

Existing index.html backed up to /opt/meshing-around/etc/www/index_backup_2025-07-28.html
Main dashboard written to /opt/meshing-around/etc/www/index.html
HTML reports generated for 2025-07-28 in /opt/meshing-around/etc/www
```

My logs however give me the following error:
```
mac = 34:B7:DA:62:84:59 not supported. Validate your config against config.template Exiting
2025-07-28 16:11:07,231 |    DEBUG | System: Initializing Interfaces
2025-07-28 16:11:07,232 | CRITICAL | System: Interface Type: ble
```

And when running the update.sh to try and fix stuff, I got the following errors:
`Failed to start pong_bot.service: Unit pong_bot.service not found.`
`Failed to start mesh_bot_reporting.service: Unit mesh_bot_reporting.service not found.`
`Failed to start mesh_bot_w3.service: Unit mesh_bot_w3.service not found.`

If I try to run the `./launch.sh meshbot I also get some python errors:
```
(venv) thursten@MeshtasticPi:/opt/meshing-around $ ./launch.sh meshbot
Traceback (most recent call last):
  File "/opt/meshing-around/mesh_bot.py", line 14, in <module>
    from modules.log import *
  File "/opt/meshing-around/modules/log.py", line 72, in <module>
    file_handler_sys = TimedRotatingFileHandler('logs/meshbot.log', when='midnight', backupCount=log_backup_count, encoding='utf-8')
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/handlers.py", line 214, in __init__
    BaseRotatingHandler.__init__(self, filename, 'a', encoding=encoding,
  File "/usr/lib/python3.11/logging/handlers.py", line 58, in __init__
    logging.FileHandler.__init__(self, filename, mode=mode,
  File "/usr/lib/python3.11/logging/__init__.py", line 1181, in __init__
    StreamHandler.__init__(self, self._open())
                                 ^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 1213, in _open
    return open_func(self.baseFilename, self.mode,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/opt/meshing-around/logs/meshbot.log'
```

I'm not sure if I also need to follow the docker steps instead of running it in the Python venv.

I'm sorry it's mostly a Info dump, if you need any extra info let me know.

I also am not sure how to connect to a chat or any other configs of the Heltec Device.

Thanks in advance.

ThirstyThursten, 2025-07-28:
> Did you install first?
> 
> Readme.md has some steps a few major ways
> 
> Install.sh will guide you through as well.
> 
> Can't generate the html till the bot runs a little.

Hey man, thanks for reaching out!

Yeah so what I did basically is do the git clone, than ran the install.sh, it asked to install in /op/meshing-around/ so I did.
went through all the questions in the script. and it all seemed fine.

However I'm kind of lost as to how to actually connect my Heltec and than have the bot do stuff, like how do I connect from my other Meshtastic devices to the Heltec and Bot on the Pi?
Say from a Lilygo T-deck Plus for example.

I almost feel like I missed some installation steps with the amount of errors I got above.

Just to be clear, you can use the ./install.sh *OR* the docker right? It's not *AND* docker. cause I didn't do that.

closed 2025-08-06 by SpudGunMan

SpudGunMan, 2025-08-12:
$whoami is either meshbot or the local current user depending on the install choice.